intel/zephyr.js #736: [ashell] Not supported special character by "mv" command

cuiyanx, 2017-02-15:
When using ashell command "mv" as rename file, the name of this file will not be supported by special character, like this "-".

Faild:
`mv temp.dat test-timer.js`

Success:
`mv temp.dat testTimer.js`

grgustaf, 2017-02-28:
We should investigate what the allowed characters really are and support all of them. It's a FAT filesystem so I believe we've already realized filenames need to be in 8.3 format (e.g. thislong.txt). It's possible that @cuiyanx really ran into a problem because the filename was too long, not because of the hyphen character.

cuiyanx, 2017-03-01:
As you say, I test that again, you are right.
Currently version: 
Faild: `mv temp.dat testTimer.js`
Success: `mv temp.dat test-la.js`
So, thanks to know and closed.

closed 2017-03-01 by cuiyanx