intel/zephyr.js #737: [ashell] Connected message is disappeared

cuiyanx, 2017-02-15:
When I open `http://localhost:8000` and connect to USB device (a101), the connected message is disappeared. It will be like this:
```
Zephyr.js DEV MODE Feb 15 2017 02:34:35
'A Shell' bash

Commands list:
    help This help
    eval Evaluate JavaScript in real-time
   clear Clear the terminal screen
    load [FILE] Saves the input text into a file
     run [FILE] Runs the JavaScript program in the file
    stop Stops current JavaScript execution
      ls [FILE] List directory contents or file stat
     cat [FILE] Print the file contents on the stdout
      du [FILE] Estimate file space usage
      rm [FILE] Remove file or directory
      mv [SOURCE] [DEST] Move a file to destination
     set Sets the input mode for 'load' accept data
        transfer raw
        transfer ihex
  reboot Reboots the device
```

grgustaf, 2017-02-28:
@brianjjones please look into this and see if you can reproduce and clarify the issue.

cuiyanx, 2017-03-06:
Fixed and closed!

closed 2017-03-06 by cuiyanx