intel/zephyr.js #745: [ashell] Hangs too easily when uploading the JS code multiple times

poussa, 2017-02-16:
When you upload the (same or different) JS application multiple times, the ashell hangs quite easily. First run is always golden, but then 2-5 run very often hangs it.

For example, the following app works always first time but hang on 2 or 3 time of running.

```js
var pins = require("arduino101_pins");
var gpio = require("gpio");

var button = gpio.open({ pin: pins.IO2, direction: 'in', edge: 'rising' });

button.onchange = function(event) {
    console.log("event: " + event.value);
}
```

qiaojingx, 2017-02-22:
Upload [test-a101-pins.js](https://github.com/01org/zephyr.js/blob/master/tests/test-a101-pins.js) multiple times. ashell will hang.
Commit: 7c5ff3c

qiaojingx, 2017-02-28:
Upload the code mentioned in description and [test-a101-pins.js](https://github.com/01org/zephyr.js/blob/master/tests/test-a101-pins.js) for 5 times individually. This issue is not reproducible.
Commit: 76c4663

brianjjones, 2017-02-28:
With the latest changes this should be fixed.  Let me know if you still see it.

grgustaf, 2017-02-28:
This was fixed by #742 and #755. @poussa please confirm and close.

grgustaf, 2017-03-21:
I think QA confirmed, closing.

closed 2017-03-21 by grgustaf