intel/zephyr.js #746: ashell: JS eval error messages are not terminated with a newline character

siovene, 2017-02-16:
All messages that I receive from ashell are terminated with CRLF, but when I try to run some JS code that has errors in it, the messages I receive are only terminated with CR, and no LF.

Because of that, my terminal will perform a CR and then overwrite the same line with the next message, garbling the text.

poussa, 2017-02-16:
This need bit more planning. That is, we need to separate at least the following data types inside the Web USB connection.

1. Control messages between the IDE and ashell (ACK, RUN, etc)
2. JavaScript console messages (via console.log)
3. JavaScript engine error messages (e.g. Syntax error in line 7, etc)

We need a protocol to make this robust and easy to extend.

brianjjones, 2017-04-05:
Until we get around to the newer protocol, I added the newlines. https://github.com/01org/zephyr.js/pull/972

grgustaf, 2018-01-24:
@brianjjones please check status.