intel/zephyr.js #757: [REG] Unable to allocate buffer on Arduino 101

qiaojingx, 2017-02-21:
### Issue Description
Unable to allocate buffer on Arduino 101

### Test Code
[HeartRateDemo.js](https://github.com/01org/zephyr.js/blob/master/samples/HeartRateDemo.js)
[WebBluetoothDemo.js](https://github.com/01org/zephyr.js/blob/master/samples/WebBluetoothDemo.js)
[WebBluetoothGroveLcdDemo.js](https://github.com/01org/zephyr.js/blob/master/samples/WebBluetoothGroveLcdDemo.js)

### Actual Result
"[ERROR]Unable to allocate buffer" will be outputted after running these samples for a while.
Detail info please refer to serial logs.

### Expected Result
No error message come out.

### Test Builds
Branch |Commit Id |Target Device |Test Date |Result
---|---|---|---|---
master|3f8eb85| Arduino 101 | Feb 13, 2017 | Pass
master|7c5ff3c| Arduino 101 | Feb 13, 2017 | Fail

### Additional Information
Serial log for: [WebBluetoothGroveLcdDemo.js](https://github.com/01org/zephyr.js/blob/master/samples/WebBluetoothGroveLcdDemo.js)
```
Boot timestamp: 60 

ipm_console0: 'Sensor core running ZJS ARC support image'
WebBluetooth Demo with Grove LCD...
Advertising as Physical Web device
Client connected: cc:fa:00:73:4f:dc (public)
Temperature change: 21C
Temperature change: 23C
Temperature change: 24C
Temperature change: 23C
Temperature change: 24C
Temperature change: 23C

src/zjs_buffer.c:347 zjs_buffer_create():
[ERROR] unable to allocate buffer

src/zjs_util.c:368 zjs_print_error_message():
[ERROR] Uncaught exception: Error: 
Temperature change: 24C

src/zjs_buffer.c:347 zjs_buffer_create():
[ERROR] unable to allocate buffer

src/zjs_util.c:368 zjs_print_error_message():
[ERROR] Uncaught exception: Error: 
Temperature change: 23C
Temperature change: 24C
```
Serial log for: [WebBluetoothDemo.js](https://github.com/01org/zephyr.js/blob/master/samples/WebBluetoothDemo.js)
```
ipm_console0: 'Sensor core running ZJS ARC support image'

WebBluetooth Demo with LED Bulb...
Advertising as Physical Web device
Client connected: cc:fa:00:73:4f:dc (public)
Temperature change: 24C
Temperature change: 26C
Temperature change: 28C
Color change: #fff000
Temperature change: 27C
Temperature change: 28C
Temperature change: 27C
Temperature change: 28C
Temperature change: 27C
Temperature change: 25C
Temperature change: 27C
Temperature change: 28C
Temperature change: 27C

src/zjs_buffer.c:347 zjs_buffer_create():
[ERROR] unable to allocate buffer

src/zjs_util.c:234 zjs_alloc_from_jstring():
[ERROR] allocation failed (1 bytes)

src/zjs_util.c:371 zjs_print_error_message():
[ERROR] Uncaught exception: Error
Temperature change: 28C

src/zjs_buffer.c:347 zjs_buffer_create():
[ERROR] unable to allocate buffer
```
Serial log for: [HeartRateDemo.js](https://github.com/01org/zephyr.js/blob/master/samples/HeartRateDemo.js)
```
ipm_console0: 'Sensor core running ZJS ARC support image'
BLE state: poweredOn
advertisingStart: success

src/zjs_event.c:444 zjs_trigger_event():
[ERROR] event object not found

src/zjs_event.c:444 zjs_trigger_event():
[ERROR] event object not found

src/zjs_event.c:444 zjs_trigger_event():
[ERROR] event object not found
Measured 1st beat
Measured 2nd beat
Subscribed to heart rate change.
Measured 1st beat
Measured 2nd beat
Measured 1st beat
Measured 2nd beat

src/zjs_util.c:234 zjs_alloc_from_jstring():
[ERROR] allocation failed (11 bytes)

src/zjs_util.c:234 zjs_alloc_from_jstring():
[ERROR] allocation failed (39 bytes)

src/zjs_util.c:371 zjs_print_error_message():
[ERROR] Uncaught exception: Error
```

grgustaf, 2017-03-21:
The problem here was that many buffers had been created and needed to be garbage collected. The JS Buffer objects held native pointers that couldn't be freed until the GC happened. #849 should solve this: it runs the JerryScript garbage collector whenever a malloc fails, and then retries the malloc.

Also, #845 will make this way less likely because now we can allocated hundreds of small buffers where we could only do a handful before. So we're much less likely to have malloc fail in the first place.

grgustaf, 2017-03-24:
@qiaojingx, @haoxli, please try to verify.

qiaojingx, 2017-03-27:
Verified with commit bb11854. This issue is not reproducible,
Closed

closed 2017-03-27 by qiaojingx