intel/zephyr.js #758: [udp] Throw our error with building UDP server in qemu

cuiyanx, 2017-02-22:
I am following the instructions on [Networking with Qemu](https://www.zephyrproject.org/doc/subsystems/networking/qemu_setup.html) to build [/samples/UDPEchoServ4.js](https://github.com/01org/zephyr.js/blob/master/samples/UDPEchoServ4.js) and [/samples/UDPEchoServ6.js](https://github.com/01org/zephyr.js/blob/master/samples/UDPEchoServ6.js), sometimes I have got some error messages.

command:
`make BOARD=qemu_x86 qemu JS=sample/UDPEchoServ4.js`

error message:
[image]
[image]

haoxli, 2017-02-22:
I also meet another case when using the same command: 
[image]

pfalcon, 2017-02-23:
> I also meet another case when using the same command:

Let me start with this one. So, there reason for that could be not following exactly the QEMU networking setup as described (linked) from a sample header itself: https://github.com/01org/zephyr.js/blob/master/samples/UDPEchoServ4.js#L6

For example, if I don't run ./loop-socat.sh or ./loop-slip-tap.sh scripts, I get:

~~~
To exit from QEMU enter: 'CTRL+a, x'
[QEMU] CPU: qemu32
qemu-system-i386: -serial unix:/tmp/slip.sock: Failed to connect socket: No such file or directory
qemu-system-i386: -serial unix:/tmp/slip.sock: could not connect serial device to character backend 'unix:/tmp/slip.sock'
~~~

Note that it's a different error message than reported above, but I may only imagine it mat be because of not following the instructions mentioned above exactly.

So, getting errors like that is not ideal - they aren't helpful to an end user. But that's how Zephyr currently works (well, the error message comes from QEMU). It would be nice to be more user-friendly, but as I said few times already, I don't have an exact idea where and how to resolve that, we need to think about it all together.

pfalcon, 2017-02-23:
Now to the first error reported - a fatal exception.

> sometimes I have got some error messages.

I can only +1 that. I sometimes get similar things from QEMU too, and that happened with different examples (e.g. with Zephyr's own echo_server, etc.). It's unclear where the issue lies. It can be Zephyr, it can be even QEMU. Well, could be my code, but it's a thin lyer on top of Zephyr API calls. The problem is that it's not reproducible. If you can reproduce it with high probability, please give the instructions I'll be happy to look into it, or we can report it upstream.

cuiyanx, 2017-02-24:
Please add doc to how to use it, that be more user-friendly.

pfalcon, 2017-03-01:
Ok, just an extra confirmation that I see these. I didn't see these crashes in a while, and now I got crash - uncrash - crash. Regdump from the latest one:

~~~
To exit from QEMU enter: 'CTRL+a, x'
[QEMU] CPU: qemu32
qemu-system-i386: warning: Unknown firmware file in legacy mode: genroms/multiboot.bin

qemu: fatal: Trying to execute code outside RAM or ROM at 0xc0020094

EAX=0100407d EBX=0040f000 ECX=004084f7 EDX=f000e2c3
ESI=ffffff86 EDI=00000003 EBP=0100407e ESP=00ffffcc
EIP=c0020094 EFL=00000213 [----A-C] CPL=0 II=0 A20=1 SMM=0 HLT=0
ES =0010 00000000 ffffffff 00cf9300 DPL=0 DS   [-WA]
CS =0008 00000000 ffffffff 00cf9b00 DPL=0 CS32 [-RA]
SS =0010 00000000 ffffffff 00cf9300 DPL=0 DS   [-WA]
DS =0010 00000000 ffffffff 00cf9300 DPL=0 DS   [-WA]
FS =0010 00000000 ffffffff 00cf9300 DPL=0 DS   [-WA]
GS =0010 00000000 ffffffff 00cf9300 DPL=0 DS   [-WA]
LDT=0000 00000000 0000ffff 00008200 DPL=0 LDT
TR =0000 00000000 0000ffff 00008b00 DPL=0 TSS32-busy
GDT=     00100080 00000017
IDT=     00121ea0 000007ff
CR0=00000033 CR2=00000000 CR3=00000000 CR4=00000000
DR0=00000000 DR1=00000000 DR2=00000000 DR3=00000000 
DR6=ffff0ff0 DR7=00000400
CCS=00000011 CCD=01004140 CCO=EFLAGS  
EFER=0000000000000000
FCW=037f FSW=0000 [ST=0] FTW=00 MXCSR=00001f80
FPR0=0000000000000000 0000 FPR1=0000000000000000 0000
FPR2=0000000000000000 0000 FPR3=0000000000000000 0000
FPR4=0000000000000000 0000 FPR5=0000000000000000 0000
FPR6=0000000000000000 0000 FPR7=0000000000000000 0000
XMM00=00000000000000000000000000000000 XMM01=00000000000000000000000000000000
XMM02=00000000000000000000000000000000 XMM03=00000000000000000000000000000000
XMM04=00000000000000000000000000000000 XMM05=00000000000000000000000000000000
XMM06=00000000000000000000000000000000 XMM07=00000000000000000000000000000000
~~~

Just as above, it's "Trying to execute code outside RAM or ROM", though address is different. And yes, I got this with UDP code, but I simply don't run anything else now. Previously, I believe saw similar with other scripts, and definitely saw similar with native Zephyr samples.

pfalcon, 2017-03-01:
Now I instead got:

~~~
***** CPU exception 6
Current thread ID = 0x004077e0
Faulting segment:address = 0x8:0x408150
eax: 0x610fae7c, ebx: 0x407c1c, ecx: 0x4, edx: 0xf000e2c3
esi: 0xe001415d, edi: 0xf0005e9a, ebp: 0407c58, esp: 0x407c00
eflags: 0xa03
Fatal fault in thread 0x004077e0! Aborting.
~~~

So, if you guys don't see these with other JS script, then well, I'd tend to agree that this is related to "dgram" module, but first relation would networking enabled in Zephyr. Though UDP networking is also enabled for OCF, I wonder if you ever see such with it (I ran it 4-5 times at the Portland Hackathon, so don't have representative sample set). If only it was reproducible... But it's completely random.

So, if you guys see any case like that, please post the crash dump here (textual paste is definitely better than screenshot), let's see if we find any pattern.

pfalcon, 2017-03-01:
(Damn, maybe I don't initialize memory somewhere? I'll re-review stuff when I get a chance, in bag-packing mode now.)

jprestwo, 2017-03-01:
@pfalcon,

Is this on QEMU? If yes I wouldn't spend too much time on it. These things happen on QEMU and I was never able to explain them, specifically with networking. It even happens with the Zephyr samples.

pfalcon, 2017-03-01:
@jprestwo : Yes, QEMU, per the title. Well, as the ticket is still open, I just add info to it. And would be nice to figure out the issue eventually ;-).

grgustaf, 2018-01-11:
I'm not able to reproduce this today with a number of tries. One thing I noticed is that I hadn't updated my "net-tools" repo since I initially cloned it last March. There have been a number of patches in there, perhaps this matters?

Can you see if you're able to reproduce, @cuiyanx?

cuiyanx, 2018-01-12:
@grgustaf No, I can't reproduce.
So, close this issue.

closed 2018-01-12 by cuiyanx