intel/zephyr.js #759: Initialization error of OCF client

qiaojingx, 2017-02-22:
When running OCF client samples ([OcfClient.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfClient.js), [OcfClientUpdate.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfClientUpdate.js), [OcfDeviceInfo.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfDeviceInfo.js), [OcfPlatformInfo.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfPlatformInfo.js)) on Arduino 101, FRDM-K64F and Linux platform with commit 7c5ff3c.

Expected result:  Initialization of OCF client succeed.

Actually result: Initialization of OCF client failed.

Serial log for Arduino 101 and Linux:
```
oc_main: stack initialized
Started OCF client

src/main.c:228 main():
[ERROR] Error running javascript

src/zjs_util.c:368 zjs_print_error_message():
[ERROR] Uncaught exception: TypeError: Base object cannot be null or undefined.
```
Nothing is outputted when running OCF on FRDM-K64F.

Branch | Target Device | Build ID | Date | Result
---|---|---|---|---
master | Arduino | 7c5ff3c | Feb 22, 2017 | FAIL
master | FRDM-K64F | 7c5ff3c | Feb 22, 2017 | FAIL
master | Linux | 7c5ff3c | Feb 22, 2017 | FAIL

jprestwo, 2017-03-03:
@cuiyanx, a fix for this is now in master.

qiaojingx, 2017-03-09:
Verified with commit b29dbbc. No error when initialize OCF client on arduino 101 and Linux.
On FRDM-K64F, only got message "Started OCF server" or "Started OCF client" when runing ocf samples.

qiaojingx, 2017-03-14:
Closed since no error thrown when initialize OCF client on arduino 101 and Linux

closed 2017-03-14 by qiaojingx