intel/zephyr.js #761: [udp] Cannot bind other ip except '192.0.2.1' and '2001:db8::1'

cuiyanx, 2017-02-23:
**JS code:**
IPv4:
```
var HOST = "192.0.2.10";
var PORT = 33333;
server.bind(PORT, HOST);
```
IPv6:
```
var HOST = "2001:db8::10";
var PORT = 33333;
server.bind(PORT, HOST);
```

**error message:**

[image]

pfalcon, 2017-02-23:
That's expected. You can bind only to an address which belongs to one of local interfaces (because you bind to a local interface(s) in the first place), and addresses you have in the title are currently hardcoded addresses for "dgram" modules (them being hardcoded is another known issue, and it's again not clear how to make them configurable, requires more discussion and then work on that).

cuiyanx, 2017-02-24:
OK, thanks to know.

closed 2017-02-24 by cuiyanx

haoxli, 2017-02-24:
@pfalcon if it's hardcoded, it's better to update the API description in [dgram.md](https://github.com/01org/zephyr.js/blob/master/docs/dgram.md)

> void bind(int port, string ip_addr);
>
> Bind socket to a local address and port. This is required operation for server-side sockets, i.e. sockets which wait and receive data from other network nodes. ip_addr must be a string representing an IP address. This module does not support domain name resolution, **so only IP addresses are allowed. An example of IPv4 address: '192.0.2.1', IPV6: '2001:db8::1'.**

From my understanding, **'192.0.2.1'** just is an example for IPv4, all other IP addresses should be allowed. How do you think of updating it to a notice like ` Now only '192.0.2.1' is available for IPv4 address, and '2001:db8::1' for IPV6` ?

pfalcon, 2017-02-27:
> all other IP addresses should be allowed.

Indeed, they are allowed - as long as a users follows (basic, well-known (?)) principles of how bind works. Perhaps, docs should provide hints of these principles though.

So, I submitted https://github.com/01org/zephyr.js/issues/765 regarding how to handle networking config in general, and will ponder how to unconfuse these matters on the levels of API, examples, and docs, patches will follow.