intel/zephyr.js #762: [ashell] "set transfer ihex/raw" is not working as expected

qiaojingx, 2017-02-24:
### Description
"set transfer ihex/raw" is not working as expected
 
### Test Code
[ashell.md](https://github.com/01org/zephyr.js/blob/master/docs/ashell.md)
 
### Steps to Reproduction
1. Set 'CONFIG_USB_CDC_ACM=y' in fragments/prj.conf.arduino_101_dev to enable USB CDC ACM device class driver.
2. Execute "make DEV=ashell" in the latest zephyrJS build
3. Execute "make dfu" to flash ashell to Arduino101 and Reset device
4. Use the following command to connect to the ashell from a terminal `$  screen /dev/ttyACM0 115200`
5. Input `acm> set transfer raw` and Enter
6. Input `acm> set transfer ihex` and Enter
### Actual Result
After 4. 
[image]

After 5.
[image]
 

### Expected Result
After 4. Transfer to RAW. 
acm> set transfer raw
RAW>

After 5. Transfer to IHEX. 
acm> set transfer ihex
HEX>

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 2e8c582| Arduino 101 | Feb 13, 2017 | Fail
 
### Additional Information

Test ashell with zephyrIDE, this issue is also reproduced.But nothing was returned after executing `set transfer ihex`.
[image]

xiaoyu3x, 2017-10-23:
hi, @qiaojingx 
According to latest [ashell](https://github.com/01org/zephyr.js/blob/master/docs/ashell.md#commands) doc, 'set transfer raw' and 'set transfer ihex' cmd have been removed, so closed it please.

qiaojingx, 2017-10-26:
Close since this command has been removed from [ashell.md](https://github.com/01org/zephyr.js/blob/master/docs/ashell.md).

closed 2017-10-26 by qiaojingx