intel/zephyr.js #764: [ocf] Can not find /a/light resource

cuiyanx, 2017-02-27:
### Issue Description

I am following the instructions on [ocf-ble.md](https://github.com/01org/zephyr.js/blob/master/docs/ocf-ble.md) to build OCF Server and Client. I have got some error message and can not find `/a/light `resource.

### Test Code

[samples/OcfServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfServer.js)

### Steps to Reproduction

1. Build [samples/OcfServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfServer.js) on a101 as OCF Server.
2. Connect Linux to a101.
3. Build [iot-rest-api-server](https://github.com/01org/iot-rest-api-server) as OCF Client.
4. Find resource with `./oic-get /res`.

### Actual Result

Can not find `/a/light` resource

### Expected Result

Find `/a/light` resource

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 7c5ff3c1 | Arduino 101 | Feb 25, 2017 | Fail

### Additional Information

[image]
[image]
[image]
[image]

jprestwo, 2017-03-01:
@cuiyanx,

I have seen this behavior before. It always seems to be related to the 6lowpan Linux driver (or so we guess). I just tried on master and everything worked as expected. Sometimes the Linux box can get into a bad state, and the only way to fix it is to reboot or remove the 6lowpan driver and start the BLE connection steps over again. If you could try again with a fresh reboot that would be great.

cuiyanx, 2017-03-10:
@jprestwo thanks for your answer.

I build OCF module following those steps:
```
$ git clone https://gerrit.zephyrproject.org/r/zephyr zephyr-project
$ cd zephyr-project
$ source zephyr-env.sh
$ cd samples/bluetooth/hci_uart
$ make BOARD=arduino_101_ble
$ dfu-util -a ble_core -D outdir/arduino_101_ble/zephyr.bin
```
here: [update BT firmware](https://www.zephyrproject.org/doc/boards/x86/arduino_101/doc/board.html#arduino-101)

closed 2017-03-10 by cuiyanx