intel/zephyr.js #766: RFC: Handling network host name (DNS) resolution

pfalcon, 2017-02-27:
This is another offshot of https://github.com/01org/zephyr.js/pull/659, similar to https://github.com/01org/zephyr.js/issues/765 . So, currently "dgram" module doesn't support domain (DNS) names, only IP addresses, and this ticket is to consider how to add such support.

There're basically 2 ways:

1. Node.js of course allows to pass a host name directly to any function expecting an address.
2. Alternatively, can provide a separate function to resolve a host name, and keep accepting only IP addresses in networking function.
3. Can do both.

Choice 1) is the easiest for user, while 2) is more flexible. And 3) of course costs more code size and time to develop.

grgustaf, 2017-02-28:
Taking a quick look, Zephyr apparently provides a sync dns_resolve() call that can so 'A' or 'AAAA' requests. So the most straightforward thing to do would probably be implement Node's DNS module w/ resolve4() and resolve6() methods. So... option 2 for starters I guess.

Supporting #1 sounds complex because the function could then block for so long resolving, and have new errors possible when resolution fails, etc.

pfalcon, 2017-03-13:
@grgustaf : Thanks for the response, definitely sounds good to go for no.1 approach. But reading https://nodejs.org/api/dns.html, I'd say dns.lookup() is what should be implemented. Any reason why you mention resolve4()/resolve6() instead?

I'm just back from another conference (so sorry for delay with reply), and taking some time off, so probably nothing will surface until April, but this one is definitely the next in my queue (together with elaborating error handling in "dgram" to be more Node.js-esque). (No commitment to MQTT so far, but I'd like to do it, as we discussed).