intel/zephyr.js #782: [ashell] Cannot import custom module

qiaojingx, 2017-03-02:
### Description

 Cannot import custom module in ashell

### Test Code

[I2CBMP280.js](https://github.com/01org/zephyr.js/blob/master/samples/I2CBMP280.js); [test-event.js](https://github.com/01org/zephyr.js/blob/master/tests/test-event.js); [test-accelerometer.js](https://github.com/01org/zephyr.js/blob/master/tests/test-accelerometer.js); [test-ambientlight.js](https://github.com/01org/zephyr.js/blob/master/tests/test-ambientlight.js); [test-gyroscope.js](https://github.com/01org/zephyr.js/blob/master/tests/test-gyroscope.js)

### Steps to Reproduction

1. Set 'CONFIG_USB_CDC_ACM=y' in fragments/prj.conf.arduino_101_dev to enable USB CDC ACM device class driver.
2. Execute "make DEV=ashell" in the latest zephyrJS build
3. Execute "make dfu" to flash ashell to Arduino101 and Reset device
4. Use the following command to connect to the ashell from a terminal $ screen /dev/ttyACM0 115200
5. Input `acm> load test.js` and Enter
6. Input `RAW> /*Paste code for test cases*/ `and CTRL+Z to finish transfer
7. Input `acm> run test.js`

### Actual Result

After 7: Got error prompt: `NotSupportedError: native_require_handler: module not found` 
[image]
### Expected Result

After 7: Test case is executed successfully.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 06004c9| Arduino 101 | Mar 02, 2017 | Fail
 
### Additional Information

This issue is also reproduced when connected with zephyrIDE
[image]

grgustaf, 2017-03-02:
@brianjjones, did this ever work? I did change the error to "NotSupportedError" recently but that shouldn't have affected functionality, I hope.

brianjjones, 2017-03-03:
@grgustaf This has never worked as it is the job of analyzer.sh to append the required JS files.  Since ashell doesn't give analyzer.sh a JS file to look at during build time, this won't work at the moment.  I'll need to do some thinking on how this could be supported in ashell.

jprestwo, 2017-03-03:
The only way to do something like this would be to have an "add module" command. And even then we would have to completely re-do our module system to dynamically add in modules, since they are all just appended to the same script currently.

brianjjones, 2017-03-14:
I will work on adding a 'save' button to the IDE, which in turn will allow you to save a module to your file system.  Ashell when running a script will then need to search for any 'require' lines and check if those files are present on the file system, and if so append them to the javascript it intends to parse and run.

brianjjones, 2017-03-29:
@grgustaf I would suggest taking this one off 0.3.  I might have it done by then, but there is a high likelihood of unintended consequences.

brianjjones, 2017-03-31:
Alright I got this working, you can see the PR here https://github.com/01org/zephyr.js/pull/959

Feedback welcome!

qiaojingx, 2017-04-07:
Will verify it after pr merged.

qiaojingx, 2017-04-18:
Verification steps:
1. Download [Assert.js](https://github.com/01org/zephyr.js/blob/master/modules/Assert.js) module from GitHub in WebIDE. (https://01org.github.io/zephyrjs-ide/#/editor)
2. Save [Assert.js](https://github.com/01org/zephyr.js/blob/master/modules/Assert.js) module.
3. Download [test-buffer.js](https://github.com/01org/zephyr.js/blob/master/tests/test-buffer.js) from GitHub
4. Connect A101 to test-buffer.js tab.
5. Run.

Result:
Hangs issue occurred. No other error thrown.
Platform: Arduino 101 with commit 47e8aef.
               WebIDE version: v0.13.0-alpha / c60c538
[image]

brianjjones, 2017-04-20:
If you are using the 'save' button on the IDE, that won't work.  Currently that just saves Assert.js to local storage on your browser.  You need to save Assert.js to the filesystem on the arduino 101.  There is not a clean, easy way to do this yet.  But here's the steps to do it, sorry it's ugly.

In the console type:
load assert.js
* Now paste your code in the window. I use the middle mouse button click to do this since right click and ctrl+v don't work *
* Press ctrl+z to save the file

Now run your test and it should find assert.js.  Please let me know if you have further issues.

qiaojingx, 2017-04-21:
@brianjjones , assert.js module can be imported by following your steps. But it only works when running samples first time. If run again, error occurs, Please refer to the screen shot.
test-buffer.js
[image]
test-event.js
[image]

Platform: Arduino101
Zephyr.js:98a3ce2
Samples: [test-buffer.js](https://github.com/01org/zephyr.js/blob/master/tests/test-buffer.js); [test-event.js](https://github.com/01org/zephyr.js/blob/master/tests/test-event.js)
Module: [Assert.js](https://github.com/01org/zephyr.js/blob/master/modules/Assert.js)

haoxli, 2017-04-25:
@brianjjones Do you have plan to fix this issue in 0.3 release? or left it as a 0.3 know issue.

qiaojingx, 2017-04-25:
Verified with the latest commit(93d58fa) in master. The issue which mentioned in my previous comment is not reproducible.

brianjjones, 2017-04-27:
@haoxli Unless @grgustaf wants it back ported, it won't be in the 0.3 release.

@qiaojingx Yes I put in a fix for this and it was merged into master, good to see it worked.

qiaojingx, 2017-07-19:
Verified with commit af7eadd in master. JS modules can be required in ashell and WebIDE. Closed

closed 2017-07-19 by qiaojingx