intel/zephyr.js #783: [ashell] Not enough memory if run a large size JS file.

qiaojingx, 2017-03-02:
### Description

 Not enough memory if run a large size JS file.

### Test Code

[HeartRateDemo.js](https://github.com/01org/zephyr.js/blob/master/samples/HeartRateDemo.js); [test-grovelcd.js](https://github.com/01org/zephyr.js/blob/master/tests/test-grovelcd.js)
 
### Steps to Reproduction

1. Set 'CONFIG_USB_CDC_ACM=y' in fragments/prj.conf.arduino_101_dev to enable USB CDC ACM device class driver.
2. Execute "make DEV=ashell" in the latest zephyrJS build
3. Execute "make dfu" to flash ashell to Arduino101 and Reset device
4. Use the following command to connect to the ashell from a terminal $ screen /dev/ttyACM0 115200
5. `acm> load test.js` and Enter
6. `RAW> /*Paste code for test cases*/ `and CTRL+Z to finish transfer
7. `acm> run test.js`
8. `acm> run test.js` again after reset device.

### Actual Result

After 6: Can't paste the whole test code one time. It will stuck
After 7: Not enough memory issue occur.
[image]
After 8: Test case will  be executed successfully.

### Expected Result

After 6: Long code can be pasted one time.
After 7: A large size JS file can always be executed successfully. No memory issue occur.

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 06004c9 | Arduino 101 | Mar 2, 2017 | Fail
 
### Additional Information

grgustaf, 2017-03-02:
Well, there will always be some size where it's too big, so #7 isn't completely valid. But it is interesting that it works after reboot. That implies maybe something isn't getting completely freed, so worth looking into.

The code paste size is a separate issue. I think the right solution to that is maybe to use kermit / sz (zmodem) type stuff to send files over serial connection. Cut and paste is pretty clumsy, and no error checking.

Both issues are worth investigating.

brianjjones, 2017-03-03:
Out of curiosity, how many lines was the file that was too big?

grgustaf, 2017-03-03:
@brianjjones, it looked like maybe they were using HeartRateDemo.js and test-grovelcd.js? Did you try those?

brianjjones, 2017-03-14:
@grgustaf Upping the RAM=72 ROM=272 works for me, I can fit both of those.  There are certainly things we can do to shrink the amount of memory needed for transfers, but I think this is better addressed by the new protocol task.  Should we close this one?

grgustaf, 2017-03-21:
@brianjjones I bet with the new malloc patch, this would work even w/o the RAM=72? Could you try that?

brianjjones, 2017-04-06:
@grgustaf  Afraid it still requires the RAM be set higher to work.

qiaojingx, 2017-09-28:
Verified with latest commit cbe7e8f in master, this issue is not reproduced.
Close.

closed 2017-09-28 by qiaojingx