intel/zephyr.js #784: [ashell] Sensor doesn't work

qiaojingx, 2017-03-02:
### Description

Sensor doesn't work

### Test Code

[Accelerometer.js](https://github.com/01org/zephyr.js/blob/master/samples/Accelerometer.js); [AmbientLight.js](https://github.com/01org/zephyr.js/blob/master/samples/AmbientLight.js); [Gyroscope.js](https://github.com/01org/zephyr.js/blob/master/samples/Gyroscope.js);
### Steps to Reproduction

1. Set 'CONFIG_USB_CDC_ACM=y' in fragments/prj.conf.arduino_101_dev to enable USB CDC ACM device class driver.
2. Execute "make DEV=ashell" in the latest zephyrJS build
3. Execute "make dfu" to flash ashell to Arduino101 and Reset device
4. Use the following command to connect to the ashell from a terminal `$ screen /dev/ttyACM0 115200`
5. `acm> load test.js` and Enter
6. `RAW> /*Paste code of test cases*/ `and CTRL+Z to finish transfer
7. `acm> run test.js`

### Actual Result

After 7: Got error prompt: 
```
Accelerometer test...
Error 
```
[image]

### Expected Result

After 7: Test case is executed successfully. Sensor starts.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 06004c9| Arduino 101 | Mar 02, 2017 | Fail
 
### Additional Information

This issue is also reproduced when connected with zephyrIDE
[image]

kenchris, 2017-03-02:
Btw the spec changed. the events are not on a ```reading``` property any longer.
```
sensor.onchange = function() {
  console.log("x: " + this.x /* or sensor.x */); 
}
```

jimmy-huang, 2017-03-02:
@kenchris yeah, I saw those changes in the working one, we'll probably update to that later

jimmy-huang, 2017-03-14:
submitted patch #827

qiaojingx, 2017-03-15:
Verified with commit 9777997 + patch #827. This issue is not reproducible.

qiaojingx, 2017-03-16:
Verified with commit 8a2964e on Arduino 101. This issue has been fixed.
Closed

closed 2017-03-16 by qiaojingx