intel/zephyr.js #785: [ashell] Could not parse JS

qiaojingx, 2017-03-02:
### Description

 Could not parse JS only when run test-error.js in ashell

### Test Code

[test-error.js](https://github.com/01org/zephyr.js/blob/master/tests/test-error.js)

### Steps to Reproduction

1. Set 'CONFIG_USB_CDC_ACM=y' in fragments/prj.conf.arduino_101_dev to enable USB CDC ACM device class driver.
2. Execute "make DEV=ashell" in the latest zephyrJS build
3. Execute "make dfu" to flash ashell to Arduino101 and Reset device
4. Use the following command to connect to the ashell from a terminal `$ screen /dev/ttyACM0 115200`
5. `acm> load test.js` and Enter
6. `RAW> /*Paste code for test cases*/ `and CTRL+Z to finish transfer
7. `acm> run test.js`

### Actual Result

After 7: Got error prompt: 
```
[ERR] Could not parse JS
Error
```
[image]

### Expected Result

After 7: Test case is executed successfully.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 06004c9| Arduino 101 | Mar 02, 2017 | Fail
 
### Additional Information

Test-error works fine in zephyrIDE.
[image]

qiaojingx, 2017-03-03:
Annotated the following code, this issue will disappear.
```
expectStandardError("calling require with a long name", RangeError, function() {
    require('supercalifragilisticexpialidocious');
});
```

brianjjones, 2017-03-27:
This is due to the dreaded 80 character line limit in ashell.  It works on IDE because that uses ihex, which strips whitespace and makes sure each ihex line is less than 80 char.  I'm working on a fix for this.

brianjjones, 2017-03-28:
@grgustaf Just FYI I don't plan on this making it into 0.3.  It will be a big ashell change and I'd hate to destabilize things.

brianjjones, 2017-08-05:
This will be fixed by zoltan's ashell refactoring.

qiaojingx, 2017-08-09:
Verified with commit ca7af31 + #1431. This issue is not reproduced.

closed 2017-08-14 by jimmy-huang