intel/zephyr.js #796: Import test_promise module failed

qiaojingx, 2017-03-07:
### Description

Import test_promise module failed on Arduino 101 and FRDM-K64F. Not reproduced on Linux.
### Test Code

[test-promise.js](https://github.com/01org/zephyr.js/blob/master/tests/test-promise.js); [test-promise-stress.js](https://github.com/01org/zephyr.js/blob/master/tests/stress/test-promise-stress.js)
 
### Steps to Reproduction

1. `$ make JS=tests/test-promise.js`
2. `$ make dfu`
3. Reset device.
 
### Actual Result

[image]

 ### Expected Result

Test case should be executed successfully. No error thrown. 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | b29dbbc | Arduino 101 | Mar 7, 2017 | Fail
master | b29dbbc | FRDM-K64F | Mar 7, 2017 | Fail
master | b29dbbc | Linux | Mar 7, 2017 | Pass

### Additional Information

jprestwo, 2017-04-03:
Fixed with #963

closed 2017-04-04 by grgustaf

qiaojingx, 2017-04-05:
Verified with commit ddcbbbe. This issue is not reproducible.

PR #963 (2017-04-04): [test_promise] Enable test_promise module on A101
Fixes #796
commit e160f65f (2017-04-04): [promises] Enable test_promise module on A101 (#963)

Fixes #796.