intel/zephyr.js #797: [REG] AsyncGPIOPin.onchange and uart.on didn't work

qiaojingx, 2017-03-07:
### Description

AsyncGPIOPin.onchange and uart.on didn't work

### Test Code

[ButtonLEDsAsync.js](https://github.com/01org/zephyr.js/blob/master/samples/ButtonLEDsAsync.js); [UART.js](https://github.com/01org/zephyr.js/blob/master/samples/UART.js); [test-gpio.js](https://github.com/01org/zephyr.js/blob/master/tests/test-gpio.js); [test-uart-read-manual.js](https://github.com/01org/zephyr.js/blob/master/tests/test-uart-read-manual.js)
 
### Steps to Reproduction

1. `$ make JS=samples/UART.js`
2. `$ make dfu`
3. Reset device. 

### Actual Result

[ButtonLEDsAsync.js](https://github.com/01org/zephyr.js/blob/master/samples/ButtonLEDsAsync.js)
 Only one on board LED is always bright, not blinking.

[UART.js](https://github.com/01org/zephyr.js/blob/master/samples/UART.js)
[image]

[test-uart-read-manual.js](https://github.com/01org/zephyr.js/blob/master/tests/test-uart-read-manual.js)
[image]

[test-gpio.js](https://github.com/01org/zephyr.js/blob/master/tests/test-gpio.js)
Test stuck at the tenth case.
[image]

### Expected Result

[ButtonLEDsAsync.js](https://github.com/01org/zephyr.js/blob/master/samples/ButtonLEDsAsync.js)
Blink the two LEDs on and off together

[UART.js](https://github.com/01org/zephyr.js/blob/master/samples/UART.js) and [test-uart-read-manual.js](https://github.com/01org/zephyr.js/blob/master/tests/test-uart-read-manual.js)
No error thrown

[test-gpio.js](https://github.com/01org/zephyr.js/blob/master/tests/test-gpio.js)
All test cases are executed successfully.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 76c4663 | Arduino 101 | Mar 1, 2017 | Pass
master | b29dbbc | Arduino 101 | Mar 7, 2017 | Fail
 
### Additional Information

qiaojingx, 2017-03-13:
Verified with commit eaf24e9 on Arduino101.  AsyncGPIOPin.onchange works fine. But issue of uart.on is still reproducible.

grgustaf, 2017-03-24:
@jprestwo please look at the UART problem reported

jprestwo, 2017-03-28:
This should be fixed on master now.

qiaojingx, 2017-03-30:
Verified with commit a2ce873 on Arduino 101.
UART issue was also be fixed. Closed

closed 2017-03-30 by qiaojingx