intel/zephyr.js #799: ashell creates empty files when you make a typo

grgustaf, 2017-03-07:
If you type the wrong file name in a run, parse, or du command at the ashell command line, it will create an empty file. I think this is because the underlying Zephyr FS API creates on open. So instead we should be checking for existence first and giving an error if it doesn't exist.

The 'cat' command *does* check, however its File not found error is missing a newline so it doesn't print nicely, so that could be fixed too.

grgustaf, 2017-03-29:
@jprestwo just copied you for your filesystem knowledge. I haven't looked yet, got to be an easy fix, either ashell is always opening for write or Zephyr APIs create on read?

brianjjones, 2017-04-05:
@jprestwo I have a fix for this, so you don't need to work on it if you haven't started.  FYI.

PR #972 (2017-04-06): [ashell] All ashell commands check if the file exists
Also added newline to all the error strings to keep the terminal
output clean.

Fixes #799
commit f4b5e276 (2017-04-06): [ashell] All ashell commands check if the file exists (#972)

Also added newline to all the error strings to keep the terminal
output clean.

Fixes #799.
closed 2017-04-06 by grgustaf