intel/zephyr.js #801: [aio] Doesn't work with AIO module by 0.9.0 zephyr-SDK

cuiyanx, 2017-03-08:
### Description

I update zephyr-SDK to 0.9.0 and check it. I find the latest Zephyr SDK doesn't work with AIO module. The issue is pointed to `var aio = require("aio")` .
 

### Test Code

[/test/test-aio.js](https://github.com/01org/zephyr.js/blob/master/tests/test-aio.js)
[/test/test-a101-pins.js](https://github.com/01org/zephyr.js/blob/master/tests/test-a101-pins.js)
[/samples/AIO.js](https://github.com/01org/zephyr.js/blob/master/samples/AIO.js)
[/samples/WebBluetoothDemo.js](https://github.com/01org/zephyr.js/blob/master/samples/WebBluetoothDemo.js)
 

### Steps to Reproduction

1. Build and setup latest Zephyr SDK with https://www.zephyrproject.org/downloads.
2. Build test case with `make JS=test/test-aio.js`.
3. Flash a101 with `make dfu`.
4. Reboot a101 and watch the screen for console log.
 

### Actual Result

No console log.
 

### Expected Result

Print some console log like other test case.
 

### Test Builds

Branch | Commit Id | Target Device | Test Date | SDK | Result
---|---|---|---|---|---
master | b29dbbc | Arduino 101 | March 7, 2017 | 0.9.0 | Fail
master | b29dbbc | Arduino 101 | March 7, 2017 | 0.8.2 | Pass

 

### Additional Information

qiaojingx, 2017-03-17:
Verified with commit 6d7a58b. This issue is not reproducible.

cuiyanx, 2017-03-22:
fixed and closed!

closed 2017-03-22 by cuiyanx