intel/zephyr.js #802: [ocf] BT will disconnect when try to find resource by ./oic-get /res

qiaojingx, 2017-03-08:
### Description

BT will disconnect when try to find resource by `./oic-get /res` in [iot-rest-api-server](https://github.com/01org/iot-rest-api-server)

### Test Code

[OcfServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfServer.js); 
 
### Steps to Reproduction

1. Build [samples/OcfServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfServer.js) on a101 as OCF Server.
2. Command: `make BOARD=arduino_101 JS=samples/OcfServer.js`.
3. Connect A101 to Ubuntu 16.04.
4. Build [iot-rest-api-server](https://github.com/01org/iot-rest-api-server) as OCF Client.
5. Find resource with `./oic-get /res`.
 
### Actual Result

BT disconnected.
[image]

[image]

### Expected Result

 BT doesn't disconnect.
 Could find `/a/light` resource according to [ocf-ble.md](https://github.com/01org/zephyr.js/blob/master/docs/ocf-ble.md)

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | b29dbbc | Arduino 101 | Mar 7, 2017 | Fail
 
### Additional Information

jprestwo, 2017-03-08:
This looks very much like your BT firmware is outdated. I saw the exact same behavior on a board that I forgot to update the firmware. Could you check and let us know.

qiaojingx, 2017-03-09:
@jprestwo , yes, this issue won't occur after updating BT firmware on Arduino101. And i can also find `/a/light` resource by [iot-rest-api-server](https://github.com/01org/iot-rest-api-server). 
thanks for your answer.
Closed

closed 2017-03-09 by qiaojingx