intel/zephyr.js #803: [fs] File system module doesnot support build for FRDM-K64F

qiaojingx, 2017-03-09:
### Description

 File system module doesnot support build for FRDM-K64F

### Test Code

[FsAsync.js](https://github.com/01org/zephyr.js/blob/master/samples/FsAsync.js); [FsSync.js](https://github.com/01org/zephyr.js/blob/master/samples/FsSync.js); [test-fs.js](https://github.com/01org/zephyr.js/blob/master/tests/test-fs.js)
 
### Steps to Reproduction

 1. `$ make BOARD=frdm_k64f JS=samples/FsAsync.js`

### Actual Result

1. Failed to build FS samples for FRDM-K64F. Error message please refer to k64f_FsAync.txt
 
### Expected Result

 1. Build FS samples successfully. 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 6b7abbd | FRDM-K64F | Mar 9, 2017 | Fail
 
### Additional Information

[k64f_FsAync.txt](https://github.com/01org/zephyr.js/files/830369/k64f_FsAync.txt)

grgustaf, 2017-03-13:
@jprestwo can you find out if there is support for the flash filesystem on K64F's microSD slot? If so, yeah let's add support for that!

jprestwo, 2017-03-24:
@grgustaf, @qiaojingx,

There is no Zephyr driver for the FRDM-K64F's SD card slot so running FS tests on it wont work. Until that support is there we should close this, or mark it as wont fix or will fix later.

qiaojingx, 2017-06-12:
Verified with commit 6180fec. FS is supported on K64F now. Find another issue #1205 .
So closed this one.

closed 2017-06-12 by qiaojingx