intel/zephyr.js #808: [ocf] Improve the information for resource and platform

cuiyanx, 2017-03-10:
### Description

I test OCF module and I find the information of resource and platform is very good, but not perfect. Please complete it.
 

### Test Code

Server code:
```
var resourceInit = {
    resourcePath: "/test/OCFserver",
    resourceTypes: ["core.test"],
    interfaces: ["/oic/if/rw"],
    discoverable: true,
    observable: true,
    properties: TestProperties
}

ocf.platform = {
    manufacturerName: "TestManufacturer",
    osVersion: "16.04",
    model: "Arduino101",
    manufacturerURL: "testurl.com",
    manufacturerDate: "11-11-2011",
    platformVersion: "5.0",
    firmwareVersion: "9.0",
    supportURL: "testurl.com/support"
}
```
Client code:
```
console.log("on('platformfound'):");
console.log("    id: " + platform.id);
console.log("    Manufacturer Name: " + platform.manufacturerName);
console.log("    osVersion: " + platform.osVersion);
console.log("    model " + platform.model);
console.log("    manufacturerURL: " + platform.manufacturerURL);
console.log("    manufacturerDate: " + platform.manufacturerDate);
console.log("    platformVersion: " + platform.platformVersion);
console.log("    firmwareVersion: " + platform.firmwareVersion);
console.log("    supportURL: " + platform.supportURL);

console.log("on('retrieve'):");
console.log("    deviceId: " + resource.deviceId);
console.log("    resourcePath: " + resource.resourcePath);
console.log("    resourceTypes: " + resource.resourceTypes);
console.log("    interfaces: " + resource.interfaces);
console.log("    properties: " + resource.properties);
```

### Steps to Reproduction

1. Build OCFserver to a101.
2. Connect to a101 by BLE.
3. Build OCFclient with linux.

### Actual Result

[image]
[image]

 

### Expected Result

No undefined object.
 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 6b7abbd6 | Arduino 101 | March 10, 2017 | Fail

 

### Additional Information

jprestwo, 2017-03-30:
This should be fixed on master.

cuiyanx, 2017-03-31:
@jprestwo Please let the information of device and platform to perfect.

[image]
[image]

jprestwo, 2017-04-03:
@cuiyanx, should be fixed with https://github.com/01org/zephyr.js/pull/952

closed 2017-04-03 by grgustaf

qiaojingx, 2017-04-05:
Verified with commit ddcbbbe. OCFobject: dataModels is still undefined.
Build [test-ocfserver.js](https://github.com/01org/zephyr.js/blob/master/tests/test-ocfserver.js) on A101, run [test-ocfclient.js](https://github.com/01org/zephyr.js/blob/master/tests/test-ocfclient.js) on Linux.
Got one fail:
[image]

commit ed043173 (2017-04-03): [ocf] Improve get/platform info on client side (#952)

Fixes #808.