intel/zephyr.js #810: [ocf] The request.resource.properties data is undefined

cuiyanx, 2017-03-10:
### Description

The request.resource.properties data is undefined and can't change this data for OCFserver.
 

### Test Code

[/samples/OcfSensorServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfSensorServer.js)

 

### Actual Result

The request.resource.properties.light data is undefined.
 

### Expected Result

Can change properties data for OCFserver.
 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 6b7abbd6| Arduino 101 | March 10, 2017 | Fail

jprestwo, 2017-03-10:
@cuiyanx, is this on a 'retrieve' or 'update' event? On a 'retrieve' event a client is asking for the servers properties, so I would expect request.resource.properties... to be undefined. If you are seeing this on an 'update' event then it means that the client didn't send any properties. What client are you using against OcfSensorServer.js? If its an incompatible client i.e. one that does not send a properties.light value then its expected to see properties.light as undefined.

cuiyanx, 2017-03-13:
@jprestwo this is on 'update' event.

Simple OCF client test case:
```
console.log("Test OCF client");

var client = require("ocf").client;

function onUpdate(resource) {
    console.log("on('update'):");
    if (resource.properties != undefined) {
        console.log("    resource.property.sensor: " + resource.properties.sensor);
        console.log("    resource.property.light: " + resource.properties.light);
    } else {
        console.log("    resource.properties not found");
    }
}

function onError(error) {
    if (error.deviceId) console.log("Device error: " + error.deviceId);
}

client.on("update", onUpdate);
client.on("error", onError);

function onfound(resource) {
    setTimeout(function() {
        client.update(resource).then(function(resource) {
            resource.properties.light = true;
        }).catch(function(error) {
            console.log("on('update'): error = " + error.name);
        });
    }, 4000);
}

client.findResources({ resourceType:"core.sensor" }, onfound).then(function(resource) {
    console.log("findResources(): deviceId = " + resource.deviceId);
}).catch(function(error) {
    console.log("findResources(): error = " + error.name);
});
```
[/samples/OcfSensorServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfSensorServer.js) console log:
[image]
Simple OCF client test case console log:
[image]

jprestwo, 2017-03-30:
@cuiyanx,

This should be fixed with #947, but there is part of your test that was incorrect as well. You are trying to update resource.properties.light inside the promise callback. The promise callback for update is just a response that gives you back the updated resource, it will not change any part of the resource on the server side while inside that callback. To update properly you must set the new property object before calling update:
```
resource.properties.light = true;
client.update(resource);
```

There was still a bug in the OCF client code, where it was checking `resource.<prop>` rather than `resource.properties.<prop>` which is now fixed with #947.

cuiyanx, 2017-03-31:
@jprestwo Update by #950 and wait for #947 merged.

PR #947 (2017-03-31): [ocf] Fix client resource property on update
Fixes #810

Client update was expecting the properties to be part of
the resource object, not resource.properties as it should
be.
commit db4051e4 (2017-03-31): [ocf] Fix client resource property on update (#947)

Client update was expecting the properties to be part of
the resource object, not resource.properties as it should
be.

Fixes #810.
closed 2017-03-31 by grgustaf