intel/zephyr.js #819: [REG][udp] Fail to build UDP server in qemu

qiaojingx, 2017-03-13:
### Description

 Build UDP server failed
 
### Test Code

[UDPEchoServ4.js](https://github.com/01org/zephyr.js/blob/master/samples/UDPEchoServ4.js); [UDPEchoServ6.js](https://github.com/01org/zephyr.js/blob/master/samples/UDPEchoServ6.js); [test-udp4-server-manual.js](https://github.com/01org/zephyr.js/blob/master/tests/test-udp4-server-manual.js); [test-udp4-server.js](https://github.com/01org/zephyr.js/blob/master/tests/test-udp4-server.js); [test-udp6-server-manual.js](https://github.com/01org/zephyr.js/blob/master/tests/test-udp6-server-manual.js); [test-udp6-server.js](https://github.com/01org/zephyr.js/blob/master/tests/test-udp6-server.js)
 
### Steps to Reproduction

1. `$ make BOARD=qemu_x86 qemu JS=samples/UDPEchoServ4.js`
 
### Actual Result

[image]

### Expected Result

UDP build successfully without error.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | b29dbbc| Linux | March 7, 2017 | Pass
master | eaf24e9| Linux | March 13, 2017 | Fail
 
### Additional Information

pfalcon, 2017-03-13:
Actually, trying to build just "make BOARD=qemu_x86" with latest master (eaf24e967bd909954c20a7409be5560904f237ee) gives me such error, so it's not UDP-related.

pfalcon, 2017-03-13:
The underlying cause of this was reported as https://github.com/01org/zephyr.js/issues/826 . There can be a one-line workaround, but I think it's better to fix the issue as described in the ticket above, or such cases will reoccur.

brianjjones, 2017-03-17:
@qiaojingx This is in master now, please run the tests again to confirm that I fixed the issue.

Thanks.

qiaojingx, 2017-03-20:
Verified with commit 02d9471. Still can't build UDP server in qemu.
[image]

grgustaf, 2017-03-21:
The problem you just reported is a different bug; that has been fixed in #857.

qiaojingx, 2017-03-22:
@grgustaf, Thanks for notice that. Create a new issue #861 to tracking this new bug.
Closed this one.

closed 2017-03-22 by qiaojingx