intel/zephyr.js #823: [ashell] Web USB does not work with the Zephyr SDK 0.9

poussa, 2017-03-13:
When I build the latest master (https://github.com/01org/zephyr.js/commit/eaf24e967bd909954c20a7409be5560904f237ee) with ashell (make DEV=ashell) usign the Zephyr SDK 0.9, I don't get the USB device at all (/dev/ACM0 or /dev/cu.usbmodem0_01). When building with Zephyr SDK 0.8.2 it works fine. 

I have no idea what is going on? Can somebody else test?

@brianjjones @nagineni

nagineni, 2017-03-13:
WebUSB alone works fine, but the issue is reproducible if we build it with the ashell. Something to do with the ashell it seems.

nagineni, 2017-03-13:
Quick update: ashell with WebUSB is working for me if I disable AIO and I2C modules while building.

brianjjones, 2017-03-13:
Will test and report back.

nagineni, 2017-03-13:
I2C seems to be working after updating zephyr deps to the latest master, but AIO is still broken. I suspect this might be a zephyr upstream or SDK issue - https://jira.zephyrproject.org/browse/ZEP-1882

nagineni, 2017-03-13:
I also noticed that the issue doesn't appear if we enable debug mode by adding CONFIG_DEBUG=y in scripts/analyze.sh.

jimmy-huang, 2017-03-15:
I can reproduce this with AIO, so a bug already filed upstream, should be fixed after updating gcc from 6.2.1 to 6.3, in the mean time we can patch it with adding CONFIG_DEBUG=y if needed.

https://jira.zephyrproject.org/browse/ZEP-1882

nagineni, 2017-03-15:
Workaround is here: https://github.com/nagineni/zephyr.js/commit/e65128

jimmy-huang, 2017-03-15:
@nagineni lgtm, would you like to submit a pr? we can merge and revert it later after Zephyr fixes it.

qiaojingx, 2017-03-17:
Verified with commit 6d7a58b
Building with ashell (`make DEV=ashell`) failed using the Zephyr SDK 0.9. But, building with Zephyr SDK 0.8.2 succeed.
[image]

qiaojingx, 2017-03-24:
Verified with commit c9991f0. This issue is not reproducible.

closed 2017-03-29 by grgustaf