intel/zephyr.js #824: [ashell] Include sensors to ashell build

poussa, 2017-03-13:
Include the sensors to ashell build. Now they are left our for some reason.

Good sample to get running is: https://github.com/01org/zephyr.js/blob/master/samples/SensorBLEDemo.js

brianjjones, 2017-03-13:
Easy change to the config file, I'll do that when I get in.

poussa, 2017-03-13:
I though so also but mine was always missing the ARC part of the sensor code...

brianjjones, 2017-03-13:
Ah yeah there has been a lot added to sensors so I need to update a few things to pull in the necessary modules for ashell.  I'll get that done today.

brianjjones, 2017-03-14:
Unfortunately file system and sensor can't be used at the same time currently.  Because ashell relies on file system, there's no way to use sensor with it currently.  I'm working with the Zephyr team to figure out a solution.

jimmy-huang, 2017-03-14:
submitted patch #827

brianjjones, 2017-03-17:
I tested it and it works for me.  @poussa can you confirm and close if its fixed for you?

jimmy-huang, 2017-03-17:
just a FYI, SensorDemo.js does not work, but it is a different issue so a separate bug should be filed, but all the other 3 samples like Accelerometer.js, Gyroscope.js and AmbientLight.js does work, which address this bug.

jimmy-huang, 2017-03-28:
I think it's okay to close this, @poussa @grgustaf?

grgustaf, 2017-04-05:
The author has abandoned this issue, let's close it. ;)

closed 2017-04-05 by grgustaf