intel/zephyr.js #828: Error thrown after subscribed to bmi160 sensor change

qiaojingx, 2017-03-14:
### Description

 Error thrown after subscribed to bmi160 sensor change

### Test Code

[SensorBLEDemo.js](https://github.com/01org/zephyr.js/blob/master/samples/SensorBLEDemo.js)
 
### Steps to Reproduction

1. `$ make JS=samples/SensorBLEDemo.js`
2. `$ make dfu`
3. Reset device and connect with 'nRF connect' app.
4. Subscribed to bmi160 sensor change by 'nRF connect' app.
5. Unsubscribed and then Subscribed again
 
### Actual Result

After 4. ringbug put errors occurred.
After 5. Fatal fault occurred.
[image]
 
### Expected Result

No error thrown.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | eaf24e9| Arduino 101 | Mar 14, 2017 | Fail
 
### Additional Information

jimmy-huang, 2017-03-14:
@qiaojingx looks like it's caused by 100hz setting, can you lower both sensor to 20hz if it fixes your problem?  the latest patch #827 will revert them back as well.

qiaojingx, 2017-03-15:
Verified with patch #827, this issue will not be reproducible.

qiaojingx, 2017-03-16:
Verified with commit 8a2964e on Arduino 101. This issue has been fixed.
Closed

closed 2017-03-16 by qiaojingx