intel/zephyr.js #829: [fs] FsAsync can't start

qiaojingx, 2017-03-14:
### Description

 FsAsync can't start

### Test Code

[FsAsync.js](https://github.com/01org/zephyr.js/blob/master/samples/FsAsync.js)
 
### Steps to Reproduction

 1. `$ make JS=samples/FsAsync.js`
 2. `$ make dfu`
 3. Reset device
### Actual Result

[image]
 
### Expected Result

FsAsync sample can be executed successfully.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | eaf24e9| Arduino 101 | Mar 14, 2017 | Fail
 
### Additional Information

According to [fs.md](https://github.com/01org/zephyr.js/blob/master/docs/fs.md#introduction), the asynchronous API's can be compiled in by enabling the pre-processor define ZJS_FS_ASYNC_APIS. How can we enable it and make FsAsync work?

grgustaf, 2017-03-30:
Should be fixed by #931.

qiaojingx, 2017-03-31:
Verified with commit 380261a on Arduino 101. This issue is not reproducible. Closed

closed 2017-03-31 by qiaojingx