intel/zephyr.js #850: [REG][event] Event object not found

qiaojingx, 2017-03-21:
### Description

 Event object not found

### Test Code

[Events.js](https://github.com/01org/zephyr.js/blob/master/samples/tests/Events.js); [test-event.js](https://github.com/01org/zephyr.js/blob/master/tests/test-event.js)
 
### Steps to Reproduction

1. `make JS=samples/tests/Events.js`
2. `make dfu`
 
### Actual Result

1. On Arduino 101, FRDM-K64F and Linux: 
[image]
2. There is another issue on linux: 
[image]
 
### Expected Result

 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | eaf24e9 | Arduino 101 | Mar 14, 2017 | Pass
master | eaf24e9 | FRDM-K64F | Mar 14, 2017 | Pass
master | eaf24e9 | Linux | Mar 14, 2017 | Pass 
master | 02d9471 | Arduino 101 | Mar 21, 2017 | Fail
master | 02d9471 | FRDM-K64F | Mar 21, 2017 | Fail
master | 02d9471 | Linux | Mar 21, 2017 | Fail 
### Additional Information

baleboy, 2017-03-23:
To be fixed before releasing 0.3 (and possibly before branching)

jprestwo, 2017-03-24:
This should now be fixed in master.

qiaojingx, 2017-03-27:
Verified with commit bb11854. This issue is not reproducible,
Closed

closed 2017-03-27 by qiaojingx