intel/zephyr.js #851: [REG][buffer] buffer.writeUInt32BE and buffer.writeUInt32LE failed

qiaojingx, 2017-03-21:
### Description

buffer.writeUInt32BE and buffer.writeUInt32LE failed 

### Test Code

[test-buffer-rw.js](https://github.com/01org/zephyr.js/blob/master/tests/test-buffer-rw.js);
 
### Steps to Reproduction

1. `make JS=tests/test-buffer-rw.js`
2. `make dfu`
 
### Actual Result

1. On Arduino 101, FRDM-K64F and Linux: 
[image]

### Expected Result

No error.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | eaf24e9 | Arduino 101 | Mar 14, 2017 | Pass
master | eaf24e9 | FRDM-K64F | Mar 14, 2017 | Pass
master | eaf24e9 | Linux | Mar 14, 2017 | Pass 
master | 02d9471 | Arduino 101 | Mar 21, 2017 | Fail
master | 02d9471 | FRDM-K64F | Mar 21, 2017 | Fail
master | 02d9471 | Linux | Mar 21, 2017 | Fail 
### Additional Information

baleboy, 2017-03-23:
To be fixed before releasing 0.3 (and possibly before branching)

grgustaf, 2017-03-29:
Fixed by #934.

qiaojingx, 2017-03-30:
Verified with commit a2ce873 + #934. This issue is not reproducible.
Will close it after patch merge.

PR #934 (2017-03-30): [buffer] Fix bug in big-endian Buffer functions
A bug was introduced by recent negative numbers patch; high positive
numbers were being treated as negatives and limited to 0x7fffffff.

Also, clean up a couple other things from that patch: mistaken assumption
that we support noAssert flag in these calls, and a test that uses
unsupported calls. I left it in, commented out, because it seems
interesting once we do support those calls.

Fixes #851.
commit 0e1d353d (2017-03-30): [buffer] Fix bug in big-endian Buffer functions (#934)

A bug was introduced by recent negative numbers patch; high positive
numbers were being treated as negatives and limited to 0x7fffffff.

Also, clean up a couple other things from that patch: mistaken assumption
that we support noAssert flag in these calls, and a test that uses
unsupported calls. I left it in, commented out, because it seems
interesting once we do support those calls.

Fixes #851.
closed 2017-03-30 by jimmy-huang