intel/zephyr.js #852: simple test case failed

qiaojingx, 2017-03-21:
### Description

simple test case failed

### Test Code

[test-buffer.js](https://github.com/01org/zephyr.js/blob/master/tests/test-buffer.js); 
 
### Steps to Reproduction

1. create a simple test case as below:
[image]
2. Build this simple test case and flash to A101.
3. Thrown error.
[image]
 
### Actual Result

1. Run test code [test-buffer.js](https://github.com/01org/zephyr.js/blob/master/tests/test-buffer.js) on Arduino 101, FRDM-K64F:
[image]
 
### Expected Result

No error thrown
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 02d9471 | Arduino 101 | Mar 21, 2017 | Fail
master | 02d9471 | FRDM-K64F | Mar 21, 2017 | Fail
master | 02d9471 | Linux | Mar 21, 2017 | Pass
### Additional Information

brianjjones, 2017-03-24:
@grgustaf I don't think we support this do we?  I don't recall us having any of the Math functions.

grgustaf, 2017-03-24:
We have always left out the Math library from JS to save space. However, we do have enough room these days that we could enable it, although it is large and would limit what other modules could be used. So this is basically a feature request, not a bug.

jprestwo, 2017-03-24:
We do have an easy-ish way to enable it. You can edit `jerry_feature.profile` and comment out the `CONFIG_DISABLE_MATH_BUILTIN` line. Then if you rebuild it will include the Math object.

kenchris, 2017-04-14:
In ES2017, ** is a default operator for Math.pow, so at some point is should be supported even if the Math functions aren't

rwaldron, 2017-04-14:
@kenchris 7th Edition: ES2016\* ;) I'm curious though—what is JerryScript's roadmap? I can't find anything anywhere. I'm going to file an issue over there to see what we can learn.

rwaldron, 2017-04-14:
@kenchris https://github.com/jerryscript-project/jerryscript/issues/1735

kenchris, 2017-04-14:
Right, I meant ES2016, the release with only two new features ;)

rwaldron, 2017-04-17:
Lol, that was the best release ever :)