intel/zephyr.js #853: [REG][buffer] The value of readUInt8() is 255 not 0 when writeUInt8(-1)

qiaojingx, 2017-03-21:
### Description

The value of readUInt8() is 255 not 0 when writeUInt8(-1)

### Test Code

[test-buffer.js](https://github.com/01org/zephyr.js/blob/master/tests/test-buffer.js); 
 
### Steps to Reproduction

1. `make JS=tests/test-buffer.js`
2. `make dfu`
 
### Actual Result

1. On Arduino 101, FRDM-K64F and Linux:
[image]

### Expected Result

No error.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | eaf24e9 | Arduino 101 | Mar 14, 2017 | Pass
master | eaf24e9 | FRDM-K64F | Mar 14, 2017 | Pass
master | eaf24e9 | Linux | Mar 14, 2017 | Fail
master | 02d9471 | Arduino 101 | Mar 21, 2017 | Fail
master | 02d9471 | FRDM-K64F | Mar 21, 2017 | Fail
master | 02d9471 | Linux | Mar 21, 2017 | Fail
### Additional Information

The uncaught expection error issue is tracking here #852

qiaojingx, 2017-03-21:
Closed by mistake

haoxli, 2017-03-23:
@qiaojingx Geoff talked with me, this one is invalid check point because the document in https://nodejs.org/dist/latest-v6.x/docs/api/buffer.html#buffer_buf_writeuint8_value_offset_noassert says: "Behavior is undefined when **value** is anything other than an unsigned 8-bit integer." 
So there is no expected value, we should remove this case. Please close it.

qiaojingx, 2017-03-23:
Closed this is invalid check point.

closed 2017-03-23 by qiaojingx