intel/zephyr.js #854: [REG][uart] invalid arguments error thrown in test-uart.js

qiaojingx, 2017-03-21:
### Description

 invalid arguments error thrown in test-uart.js

### Test Code

[test-uart.js](https://github.com/01org/zephyr.js/blob/master/tests/test-uart.js); 
 
### Steps to Reproduction

1. `make JS=tests/test-uart.js`
2. `make dfu`
 
### Actual Result

`[ERROR] Uncaught exception: TypeError: invalid arguments` error thrown and test-uart.js hangs at the last test case.
[image]

### Expected Result

No error.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | eaf24e9 | Arduino 101 | Mar 14, 2017 | Pass
master | 02d9471 | Arduino 101 | Mar 21, 2017 | Fail

### Additional Information

baleboy, 2017-03-23:
To be fixed before releasing 0.3 (and possibly before branching)

jprestwo, 2017-03-27:
@qiaojingx, @baleboy,

First off, this was introduced after the ZVAL patch, so now an error is thrown if the arguments are bad, rather than the promise rejecting. This is why this issue exists. Now the question comes down to, if there is a programmatic error (bad arguments) should an error be thrown immediately, or should it be handled by the promise catch function?

@grgustaf and I discussed this and came to the conclusion that it should NOT be handled by the promise catch function and instead be thrown as an error (catchable by try/catch if required). There are good reasons on either side of the coin, but I think for our use cases we should not allow programmatic errors in our code, and instead immediately throw/exit/break the current application, which show the developer they did something wrong rather than letting it slip by. One reason would be if they never used .catch() it would just continue as if nothing went wrong.

If you agree with this, I think that last test case should be removed OR surrounded with a try/catch block. Thoughts?

grgustaf, 2017-03-29:
Working as designed currently.

cuiyanx, 2017-03-30:
Will be fix by #943

qiaojingx, 2017-03-30:
Verified with updated case. No issue occurred. Closed

closed 2017-03-30 by qiaojingx