intel/zephyr.js #863: [REG][ble] Can't start ble in SensorBLEDemo.js and HeartRateDemo.js using ZephyrSDK 0.8.2

qiaojingx, 2017-03-22:
### Description

Can't start ble in SensorBLEDemo.js and HeartRateDemo.js
 
### Test Code

[SensorBLEDemo.js](https://github.com/01org/zephyr.js/blob/master/samples/SensorBLEDemo.js); [HeartRateDemo.js](https://github.com/01org/zephyr.js/blob/master/samples/HeartRateDemo.js)

### Steps to Reproduction

1. `$ make JS=samples/SensorBLEDemo.js`
2. `make dfu`
 
### Actual Result
1. When using ZephyrSDK 0.8.2:
[image]
[image]

2. When using ZephyrSDK 0.9:
[image]
[image]

### Expected Result

Able to start ble in SensorBLEDemo.js and HeartRateDemo.js when using ZephyrSDK 0.8.2
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result | SDK version 
---|---|---|---|---|---
master | 02d9471 | Arduino 101 | March 22, 2017 | Failed | 0.8.2
master | 02d9471 | Arduino 101 | March 22, 2017 | Pass | 0.9
 
### Additional Information

baleboy, 2017-03-23:
To be fixed before releasing 0.3 (and possibly before branching)

grgustaf, 2017-03-24:
@jimmy-huang, is this specific to 0.8.2?

jimmy-huang, 2017-03-27:
@grgustaf I reverted back to 0.8.2, I can't reproduce this, can you try?  Both the SensorBLEDemo.js and HeartRateDemo.js are working

ipm_console0: 'Sensor core running ZJS ARC support image'
Sensor (Accelerometer/Gyroscope) BLE Demo...
Advertising as Physical Web device
Client connected: 5a:5e:ea:44:99:2d (random)
Subscribed to bmi160 sensor change
Unsubscribed to bmi160 sensor change
Client disconnected: 5a:5e:ea:44:99:2d (random)
Client connected: 5a:5e:ea:44:99:2d (random)
Subscribed to bmi160 sensor change
Unsubscribed to bmi160 sensor change
Client disconnected: 5a:5e:ea:44:99:2d (random)

ipm_console0: 'Sensor core running ZJS ARC support image'
BLE state: poweredOn
advertisingStart: success
Measured 1st beat
Measured 1st beat
Measured 2nd beat
ipm_console0: 'Sensor core running ZJS ARC support image'

qiaojingx, 2017-03-28:
Verified with latest commit bb11854 in master. This issue is not reproducible.
Closed

closed 2017-03-28 by qiaojingx