intel/zephyr.js #864: [REG][ashell] Build with ashell failed using the Zephyr SDK 0.9

qiaojingx, 2017-03-22:
### Description

Build with ashell failed using the Zephyr SDK 0.9
 
### Test Code

[ashell.md](https://github.com/01org/zephyr.js/blob/master/docs/ashell.md)

### Steps to Reproduction

1. `$ make DEV=ashell`
 
### Actual Result
1. ZephyrSDK 0.9:
[image]

### Expected Result

Build with ashell successfully using the Zephyr SDK 0.9
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result | SDK version 
---|---|---|---|---|---
master | 9f48297 | Arduino 101 | March 22, 2017 | Failed | 0.9
master | 9f48297 | Arduino 101 | March 22, 2017 | Pass | 0.8.2
 
### Additional Information

qiaojingx, 2017-03-23:
Verified with commit a6c321a. ashell still can't be built with Zephyr SDK 0.9.
But the error message changed:
[image]

jimmy-huang, 2017-03-23:
submitted patch #873

qiaojingx, 2017-03-23:
Verified with commit a6c321a + Patch #873. This issue is not reproduced.
Will close after patch merged.

baleboy, 2017-03-23:
To be fixed before releasing 0.3 (and possibly before branching)

qiaojingx, 2017-03-24:
Verified with commit b7d9ff7 in master. This issue is not reproducible.
Closed

closed 2017-03-24 by qiaojingx