intel/zephyr.js #870: [reg][snapshot] Invalid snapshot format error thrown

qiaojingx, 2017-03-23:
### Description

Invalid snapshot format error thrown

### Test Code

All [Samples](https://github.com/01org/zephyr.js/tree/master/samples) and [tests](https://github.com/01org/zephyr.js/tree/master/tests).
 
### Steps to Reproduction

Take [HelloWorld.js](https://github.com/01org/zephyr.js/blob/master/samples/HelloWorld.js) for an example.
1. `make JS=samples/HelloWorld.js `
2. `make dfu`

### Actual Result

 
[image]

### Expected Result

 
[image]

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 02d9471 | Arduino 101 | Mar 21, 2017 | Pass
master | a6c321a | Arduino 101 | Mar 23, 2017 | Fail
 
### Additional Information

It will work if build with command `make JS=samples/HelloWorld.js SNAPSHOT=off`

jimmy-huang, 2017-03-23:
submitted patch #871

qiaojingx, 2017-03-23:
Verified with commit a6c321a + Patch #871 . This issue is not reproduced.
Will close after patch merged.

qiaojingx, 2017-03-24:
Verified with commit b7d9ff7 in master. This issue is not reproduced.
Closed

closed 2017-03-24 by qiaojingx