intel/zephyr.js #872: [ocf] ocf server down after running about 1 min

qiaojingx, 2017-03-23:
### Description

ocf server down after running about 1 min

### Test Code

[OcfServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfServer.js); 

 ### Steps to Reproduction

1. Build OCFserver to a101.
2. Connect to a101 by BLE.
3. Build OCFclient with linux.

### Actual Result

1 min later, ocfserver:
[image]

1 min later, ocfclient: 
[image]

### Expected Result

Ocfserver always works.

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | a6c321a | Arduino 101 | Mar 23, 2017 | Fail
 
### Additional Information

jprestwo, 2017-04-03:
Fixed with https://github.com/01org/zephyr.js/pull/962

closed 2017-04-03 by grgustaf

qiaojingx, 2017-04-05:
Verified with commit ddcbbbe. This issue is not reproducible.

PR #962 (2017-04-03): [promise] Set callback ID to -1 after removing them
Fixes #872
commit f2746c34 (2017-04-03): [promise] Set callback ID to -1 after removing them (#962)

Fixes #872