intel/zephyr.js #874: [reg][ashell] ashell default connect to serial console instead of browser IDE

qiaojingx, 2017-03-23:
### Description

ashell default connect to serial console instead of browser IDE

### Test Code

[ashell.md](https://github.com/01org/zephyr.js/blob/master/docs/ashell.md)
 
### Steps to Reproduction

1. `make DEV=ashell `
2. `make dfu`
3. Reset device
### Actual Result

1. No notification from Chrome pop up.
2. ashell can be connected from terminal via `screen /dev/ttyACM0 115200`
[image]

### Expected Result

1. Ashell default connect to browser IDE. A notification from Chrome pop up when plug in device to PC.
[image]

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 02d9471 | Arduino 101 | Mar 21, 2017 | Pass
master | a6c321a | Arduino 101 | Mar 23, 2017 | Fail
 
### Additional Information

qiaojingx, 2017-03-24:
Verified with commit c9991f0. Also reproduced with Zephyr SDK 0.9.

brianjjones, 2017-03-24:
Ashell is now built using either:
make ide (builds ashell for the IDE)
make ashell (builds for the command line)

I will submit a patch that disables the old way of doing it to avoid confusion.

grgustaf, 2017-03-24:
@brianjjones actually I'd like you to check for them passing DEV=ashell and give an error explaining the new ways. If we just do nothing, it builds a101 default.

brianjjones, 2017-03-24:
@grgustaf While I like the idea short term, does it really make sense to add a bunch of errors to our Makefile explaining that old depreciated things are no longer valid?  Would it make more sense to make a catch-all error that throws if the user passes an arg we don't support? And then list the args we do support? I think that might be more sustainable in the long run.  If not, I can easily add the check no sweat.

brianjjones, 2017-03-25:
I started this PR to get the ball rolling https://github.com/01org/zephyr.js/pull/894

brianjjones, 2017-03-28:
This is now fixed, you'll get an error if you try and use DEV=ashell.

qiaojingx, 2017-03-29:
Verified with commit 1c67773. 
`make DEV=ashell` is unusable now.
`make ashell` and `make ide` work.
Closed

closed 2017-03-29 by qiaojingx