intel/zephyr.js #875: [udp] callback function of receive data works abnormally

qiaojingx, 2017-03-23:
### Description

callback function of receive data works abnormally

### Test Code

[UDPEchoServ4.js](https://github.com/01org/zephyr.js/blob/master/samples/UDPEchoServ4.js); [UDPEchoServ6.js](https://github.com/01org/zephyr.js/blob/master/samples/UDPEchoServ6.js); [test-udp4-server-manual.js](https://github.com/01org/zephyr.js/blob/master/tests/test-udp4-server-manual.js); [test-udp6-server-manual.js](https://github.com/01org/zephyr.js/blob/master/tests/test-udp6-server-manual.js); [test-udp4-server.js](https://github.com/01org/zephyr.js/blob/master/tests/test-udp4-server.js); [test-udp6-server.js](https://github.com/01org/zephyr.js/blob/master/tests/test-udp6-server.js);

 ### Steps to Reproduction

1. `make BOARD=qemu_x86 qemu JS=samples/UDPEchoServ4.js`

### Actual Result

UDP4:
[image]
[image]

UDP6:
[image]
[image]

### Expected Result

UDPserver works well.

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | a6c321a | Linux | Mar 23, 2017 | Fail
 
### Additional Information

grgustaf, 2017-03-24:
@pfalcon, maybe you could take a look at this. I did mess with dgram.c with a couple global changes recently and might have broken something. I'm afraid I didn't test every module...

pfalcon, 2017-03-24:
I will. Sorry, I have some issues, so I am, and will be, slow for some time... :-I

pfalcon, 2017-03-25:
Well, tried to look into this, and found that the latest net-tools master is broken for me: https://jira.zephyrproject.org/browse/ZEP-1965

pfalcon, 2017-03-25:
Confirmed the issue, on packet reception I get:

~~~
src/zjs_util.c:379 zjs_print_error_message():
[ERROR] Uncaught exception: Error: Cannot read property 'address' of undefined
~~~

grgustaf, 2017-03-29:
BTW I did fix one bug in dgram that was showing up in the screenshots above with PR #911. That was the "callback -40 does not exist" thing.

qiaojingx, 2017-03-30:
Verified with commit a2ce873 on linux. This issue is not reproducible. Closed.

closed 2017-03-30 by qiaojingx