intel/zephyr.js #876: [REG] [ocf] Cannot set BLE-address for OCF server

cuiyanx, 2017-03-23:
### Description

I follow the [/docs/ocf.md](https://github.com/01org/zephyr.js/blob/master/docs/ocf.md) to set BLE-address for OCF server, I am failed. I should try those:
```
ocf.setBleAddress 
server.setBleAddress
setBleAddress
```
All failed.
 

### Test Code

```
var ocf = require("ocf");
var BleAddress = "F1:E2:D3:C4:B5:A6";
ocf.setBleAddress(BleAddress);
```

 

### Steps to Reproduction

 

### Actual Result

[image]

 

### Expected Result

Set BLE address for OCF server and no error.
 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result | SDK
---|---|---|---|---|---
master | a6c321a | Arduino 101 | Mar 23, 2017 | Fail | 0.9

 

### Additional Information

jprestwo, 2017-03-27:
This should now be fixed in master.

cuiyanx, 2017-03-28:
Yes, this is fixed and closed.

closed 2017-03-28 by cuiyanx