intel/zephyr.js #886: [fs] Read data that existed in r file failed when first run after power on

qiaojingx, 2017-03-24:
### Description

 Read data that existed in r file failed when first run after power on

### Test Code

[test-fs.js](https://github.com/01org/zephyr.js/blob/master/tests/test-fs.js);
 
### Steps to Reproduction

1. `make JS=tests/test-fs.js`
2. `make dfu`
3. Reset device(if not reproduced, please power off then power on device.)

### Actual Result

 Read data that existed in r file failed when first run after power on
[image]

### Expected Result

 Read data that existed in r file always successfully.

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | b7d9ff7 | Arduino 101 | Mar 24, 2017 | Fail
 
### Additional Information

Only reproduced when first run after power on.

jprestwo, 2017-03-24:
@qiaojingx, I just tested on master (c9991f0) and did not see that failure.

@grgustaf, @brianjjones, could one of you test this really quick. (just run tests/test-fs.js).

qiaojingx, 2017-03-28:
Verified with commit bb11854 in master. Still can see this issue when first run after flash.

qiaojingx, 2017-03-30:
Verified with commit a2ce873 on Arduino 101.
This issue is not reproducible. Closed

closed 2017-03-30 by qiaojingx