intel/zephyr.js #888: [reg][ocf] Extra '\f' is added after '/a/light'

qiaojingx, 2017-03-24:
### Description

Extra '\f' is added after '/a/light'

### Test Code

[OcfServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfServer.js); [OcfNotifyServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfNotifyServer.js); [OcfSensorServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfSensorServer.js);
 
### Steps to Reproduction

1. `make BOARD=arduino_101 JS=samples/OcfServer.js`
2. `make dfu`
3. Reset device
4. Connect device to PC via BLE([ocf-ble.md](https://github.com/01org/zephyr.js/blob/master/docs/ocf-ble.md))
5. Build [iot-rest-api-server](https://github.com/01org/iot-rest-api-server) as OCF Client.
6. Find resource with ./oic-get /res.

### Actual Result

[image]

### Expected Result

[image]

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 02d9471 | Arduino 101 | Mar 21, 2017 | Pass
master | b7d9ff7 | Arduino 101 | Mar 24, 2017 | Fail
 
### Additional Information

grgustaf, 2017-03-29:
Should be fixed in #922.

closed 2017-03-29 by grgustaf

qiaojingx, 2017-04-05:
Verified with commit ddcbbbe. This issue is not reproducible.

commit 8d0d8681 (2017-03-29): [ocf] Fixed null termination bug (#922)

Fixes #888.