intel/zephyr.js #898: [ashell] No enough space for arc image when build ashell/ide by default

qiaojingx, 2017-03-27:
### Description

 No enough space for arc image when build ashell/ide by default

### Test Code

[ashell.md](https://github.com/01org/zephyr.js/blob/master/docs/ashell.md)
 
### Steps to Reproduction

1. `make ashell`  or `make ide`
### Actual Result

 [image]

### Expected Result

Build ashell and ide successfully by default 

### Test Builds

Branch | Commit Id | ROM | Test Date | Result
---|---|---|---|---
master | bb11854 | default 256| Mar 27, 2017 | Fail
master | bb11854 | ROM=250 | Mar 27, 2017 | Pass

### Additional Information

According to 'Getting more space on your Arduino 101' in [README.md](https://github.com/01org/zephyr.js/blob/master/README.md), 
Decrease ROM by passing ROM=250 to increase space for ARC.
Ashell will be built successfully by the following command:
`make ashell ROM=250` or `make ide ROM=250`

grgustaf, 2017-03-27:
This is a problem with SDK 0.9 only, because we made it build in debug mode to work around a problem with the SDK. This is why SDK 0.8.2 is still the default in Travis, I don't consider 0.9 ready for us yet.

qiaojingx, 2017-05-12:
Verified with commit 782416d using zephyr SDK 0.9. This issue is not reproducible.
`make ashell` or `make ide` will get `'ROM' overflowed by xx bytes`, that is same as on zephyr SDK 0.8.2.
`make ashell ROM=256` or `make ide ROM=256` will build successfully.
Closed

closed 2017-05-12 by qiaojingx