intel/zephyr.js #899: [sensor] Please add sensor.timestamp following W3C sensor

cuiyanx, 2017-03-27:
### Description

[W3C sensor.timestamp](https://www.w3.org/TR/generic-sensor/#sensor-timestamp):
```
8.1.3. Sensor.timestamp
The getter of the timestamp attribute returns latest reading["timestamp"]
```
Please add it for zephyr.js sensor.

### Test Code

Add console.log on [/samples/AmbientLight.js](https://github.com/01org/zephyr.js/blob/master/samples/AmbientLight.js):

`console.log("sensor.timestamp: " + sensor.timestamp);`

[image]

 

### Steps to Reproduction

 

### Actual Result

[image]

 

### Expected Result

Get the latest reading value.
 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | bb11854 | Arduino 101 | Mar 27, 2017 | Fail

 

### Additional Information

grgustaf, 2017-03-31:
Fixed in #942

cuiyanx, 2017-03-31:
Thanks, be closed

closed 2017-03-31 by cuiyanx