intel/zephyr.js #901: [REG][PnP][Zephyr][Arduino101]Zephyr_WebBluetoothGroveLcdDemo_ROM(Byte) KPI has ~6% gap compared with master commit 02d9471

hanyia, 2017-03-27:
Test with WW13 master commit: bb11854, detail result as below table:

|KPI |ww13 bb11854 | ww12 02d9471 | ww13 vs ww12
----|----|------|----
|Zephyr_WebBluetoothGroveLcdDemo_ROM(Byte) | 181144| 170520 | 6.23%

hanyia, 2017-03-28:
Update to SDK v0.9, and retest WW13 master commit: bb11854, detail result as below table:

|KPI |ww13 bb11854 | ww12 02d9471 | ww13 vs ww12
----|----|------|----
|Zephyr_WebBluetoothGroveLcdDemo_ARC_ROM| 24260 | 22840 | 6.22%
|Zephyr_WebBluetoothGroveLcdDemo_ROM(Byte) | 179522 | 170520 | 5.28%

jimmy-huang, 2017-04-05:
the arc size image is probably due to enable debug flag we added as a work-around for SDK fix, the increase on x86 could be something to do with jerryscript, because size of outdir/arduino_101/libjerry-core.a on commit bb11854913be95f8ed8694e722c62c8170c5610f is 2052946, where as the size on commit 02d94718e50e4370337c5882841c2715246b6a7e is only 1315622

jimmy-huang, 2017-04-05:
Okay, i found the cause of the increase on the x86 side, one of the previous patch https://github.com/01org/zephyr.js/commit/3d5ff82e9c67e435319a835528b0a565c30620ce, broke the Makefile and caused us to ignore the jerryscript flags we passed in, and building JerryScript without the optimized footprint, so I submitted a patch to fix this here #976, now x86 should be much smaller, on master is 169400

hanyia, 2017-04-06:
This issue had fixed on  ww14 ddcbbbe master build, so will close this bug.

PR #976 (2017-04-06): [build] Fix makefile bug that EXT_JERRY_FLAGS are ignored
One of the previous patch broke the Makefile so that
jerryscript flags are not appended to EXT_JERRY_FLAGS,
which caused final ROM size to go up significantly.

Fixes #901
commit c214dd30 (2017-04-06): [build] Fix makefile bug that EXT_JERRY_FLAGS are ignored (#976)

One of the previous patch broke the Makefile so that
jerryscript flags are not appended to EXT_JERRY_FLAGS,
which caused final ROM size to go up significantly.

Fixes #901
closed 2017-04-06 by hanyia