intel/zephyr.js #914: [sensor] Please change sensor.onchange() to sensor.onactivate() following W3C sensor

cuiyanx, 2017-03-28:
### Description

I found the function sensor.onchange() has be updated to sensor.onactivate() in W3C sensor. And we update [/docs/sensors.md](https://github.com/01org/zephyr.js/blob/master/docs/sensors.md) to sensor.onactivate(), but the samples and tests had not. So, please Add sensor.onactivate() for zephyr JS following W3C sensor.
 

### Test Code

[/modules/GenericSensor.js](https://github.com/01org/zephyr.js/blob/master/modules/GenericSensor.js)
[/samples/Gyroscope.js](https://github.com/01org/zephyr.js/blob/master/samples/Gyroscope.js)
[/samples/AmbientLight.js](https://github.com/01org/zephyr.js/blob/master/samples/AmbientLight.js)
[/samples/Accelerometer.js](https://github.com/01org/zephyr.js/blob/master/samples/Accelerometer.js)

 

### Steps to Reproduction

 

### Actual Result

Function sensor.onactivate() has undefined.
 

### Expected Result

Be defined and work well.
 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 694cf15 | Arduino 101 | Mar 28, 2017 | Fail

 

### Additional Information

jimmy-huang, 2017-03-28:
@kenchris can you confirm that onactivate() completely replaces onstatechange()?  I thought onactivate is only for 1 state event from "activating" to "activated", what about other state transitions like from "activated" to "idle"?

kenchris, 2017-04-03:
The states are still being changed a bit. Like currently it looks like the errored state is going away.

Please join the discussions on the generic sensor spec, issues 160

jimmy-huang, 2017-04-06:
I am going to downgrade this to p3 as there's no immediately need to follow the spec change until it's finalized

kenchris, 2017-04-06:
There is no event from activated to idle as it is sync apparently

closed 2017-04-28 by grgustaf

cuiyanx, 2018-01-10:
Fixed and closed

commit f86a86bb (2017-04-28): [sensor] Remove onstatechange event callback (#1079)

Updating implementation and samples due to change of latest W3C spec
no longer has onstatechange.

Fixes #914