intel/zephyr.js #916: [reg][buffer] malloc failed in buffer

qiaojingx, 2017-03-28:
### Description

malloc failed in buffer
 
### Test Code

[test-buffer.js](https://github.com/01org/zephyr.js/blob/master/tests/test-buffer.js)
 
### Steps to Reproduction

1. `make JS=tests/test-buffer.js`
2. `make dfu`
 
### Actual Result

[image]
 
### Expected Result

No error thrown
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 02d9471| Arduino 101 | Mar 23, 2017 | Pass
master | 02d9471| FRDM-K64F | Mar 23, 2017 | Pass
master | bb11854| Arduino 101 | Mar 28, 2017 | Fail
master | bb11854| FRDM-K64F | Mar 28, 2017 | Fail

### Additional Information

grgustaf, 2017-03-28:
This is actually not a bug. Obviously we can't actually create a 4GB buffer on an A101. So it should have always failed like this.

The [Node docs](https://nodejs.org/dist/latest-v6.x/docs/api/buffer.html#buffer_new_buffer_size) say that buffers over [kMaxLength](https://nodejs.org/dist/latest-v6.x/docs/api/buffer.html#buffer_buffer_kmaxlength) should throw a RangeError. On a 32-bit platform like this that should be basically 1GB. However, we only have 80K of RAM on this platform so obviously that's still ridiculous. So it doesn't really make sense to follow this part of the API. Instead I think for now the right thing to do is just throw an error when malloc fails, like we're doing. The max you can really allocate will depend on memory conditions but in the best case it would be something like 20K.

grgustaf, 2017-03-28:
We should update the test case; there's not really going to be a good "maximum" value to use, but you could check to make sure an error is thrown when you use a very large size like 1GB that is obviously outside the scope of ZJS target devices. Linux can run on systems as small as 8MB RAM, and Node.js can run on systems with as little as 64MB, so anything that big would always be out of scope for ZJS.

grgustaf, 2017-03-28:
I just submitted #921 to update the docs to be more clear on intended constructor behavior.

cuiyanx, 2017-03-30:
@grgustaf  Please throw out an error when too long or not enough available memory.
I test this, just console log, not error.

cuiyanx, 2017-03-30:
Will be fix by #944 .

grgustaf, 2017-03-31:
@cuiyanx, agreed... in PR #955 it will throw an error.

qiaojingx, 2017-04-06:
Verified with commit ddcbbbe + #966 on Arduino 101. test-buffer.js works. 
Will close it after #966 merged.

grgustaf, 2017-04-07:
Reconsidered as a request to throw an error; so removed 'invalid' label.

qiaojingx, 2017-04-07:
Verified with commit f8f9621 on Arduino and K64f. Error thrown when buffer overflow.
[image]
Closed as fixed.

closed 2017-04-07 by qiaojingx