intel/zephyr.js #917: Event object not found error thrown

qiaojingx, 2017-03-28:
### Description

Event object not found error thrown 
 
### Test Code

[test-event.js](https://github.com/01org/zephyr.js/blob/master/tests/test-event.js); [HeartRateDemo.js](https://github.com/01org/zephyr.js/blob/master/samples/HeartRateDemo.js)
 
### Steps to Reproduction

1. `make JS=samples/HeartRateDemo.js`
2. `make dfu`
3.  Reset device
4. Try to connect BLE via nRF connect app

### Actual Result

[image]

[image]

### Expected Result

No error thrown
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | bb11854| Arduino 101 | Mar 28, 2017 | Fail
master | bb11854| FRDM-K64F | Mar 28, 2017 | Fail
master | bb11854| Linux | Mar 28, 2017 | Fail

### Additional Information

jprestwo, 2017-03-30:
@cuiyanx,

This has been the same behavior since the beginning. In the test, you try calling event API's on a non event object so an error is printed. There is no error throw.

cuiyanx, 2017-03-31:
[/docs/events.md](https://github.com/01org/zephyr.js/blob/master/docs/events.md):
```
emit:
boolean emit(string event, optional arg1, ...);
Triggers an event. Any listener functions which have been added to the event emitter under the event name will be called.
The event argument is the name of the event that you want to emit.
All other arguments will be given to any registered listener functions.
Returns true if there were any listener functions called.
```
I test the last line: Returns true if there were any listener functions called.

Normal situation returns "true":
```
eventEmitter.on("event_noArg", test_listener_noArg);
EmiteventFlag = eventEmitter.emit("event_noArg");
```

Exceptional situation returns "false":
` EmitnoeventFlag = eventEmitter.emit("value");`

So, when a non-event object be called, an error information should be printed.

jprestwo, 2017-03-31:
The return value is correct, its just printing that message. I can change the ERR_PRINT to a DBG_PRINT though so it will only show up in debug mode.

closed 2017-03-31 by grgustaf

grgustaf, 2017-04-01:
@qiaojingx, @haoxli hmm when I put "Fixes #xxx" in the github comments it closes the issue automatically. Is that okay for you guys? Do you track them elsewhere where you'd be sure to still verify them? Or should we instead be careful not to do that and let you close them?

If you could maybe email the answer I might not see it otherwise. :)

haoxli, 2017-04-01:
@grgustaf this is OK for us for put “Fiixes #xxx”. We’re tracking issues in the excel and always keep an eye on open issues and closed issues from github notifications , if it’s not verified but closed, we also will add verification comments.  
For the closed issue not ready fixed, we will add comments for reopen them after verification, please pay attention to the notifications.

qiaojingx, 2017-04-01:
Verified with commit f3d587d on Arduino 101. This issue is not reproduced.

PR #953 (2017-03-31): [events] Changed ERR_PRINT to DBG_PRINT in trigger event
Fixes #917

If emit is called on an event object that has no events it
would print an error message. This was not really an "error",
just a message.
commit f3d587d1 (2017-03-31): [events] Changed ERR_PRINT to DBG_PRINT in trigger event (#953)

If emit is called on an event object that has no events it
would print an error message. This was not really an "error",
just a message.

Fixes #917.