intel/zephyr.js #918: [udp] messy code is outputted on K64F

qiaojingx, 2017-03-28:
### Description

messy code is outputted on K64F
 
### Test Code

[UDPEchoServ4.js](https://github.com/01org/zephyr.js/blob/master/samples/UDPEchoServ4.js); [UDPEchoServ6.js](https://github.com/01org/zephyr.js/blob/master/samples/UDPEchoServ6.js); [test-udp4-server-manual.js](https://github.com/01org/zephyr.js/blob/master/tests/test-udp4-server-manual.js); [test-udp6-server-manual.js](https://github.com/01org/zephyr.js/blob/master/tests/test-udp6-server-manual.js); [test-udp4-server.js](https://github.com/01org/zephyr.js/blob/master/tests/test-udp4-server.js); [test-udp6-server.js](https://github.com/01org/zephyr.js/blob/master/tests/test-udp6-server.js);
 
### Steps to Reproduction

1. `make BOARD=frdm_k64f JS=samples/UDPEchoServ4.js` 
2. cp outdir/frdm_k64f/zephyr.bin /media/<USERNAME>/MBED/

### Actual Result

[image]
[image]
[image]
[image]
 
### Expected Result

No messy code output 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | bb11854 | FRDM-K64F | Mar 28, 2017 | Fail
 
### Additional Information

pfalcon, 2017-03-28:
Note: I don't have access to frdm_k64f at this time (for few weeks), so won't be able to test this :-(.

haoxli, 2017-03-30:
#386 feature verification is blocked by this issue.

pfalcon, 2017-03-30:
So, I thought for a while what that "messy code" (garbage output) could be, and I guess I got it - that looks like SLIP being output to a (user) serial console. That reminds me of one of the decision points on critical path to further evolution of networking support in Z.js: proper compile-time (prj.conf) networking configuration. Inasmuch as I like hardware-independ config, I couldn't come up with an idea how to do such for different boards having very different network interfaces. So, in MicroPython, I reduced to having a per-BOARD config fragments to configure networking. That's the way I may suggest to Z.js too.

jprestwo, 2017-04-05:
This should now be fixed on master

qiaojingx, 2017-04-06:
Verified with commit ddcbbbe on K64F.  Messy code gone. But still can't  receive message.

closed 2017-04-06 by qiaojingx