intel/zephyr.js #919: [gpio] Fatal fault when running test-k64k-pins.js

qiaojingx, 2017-03-28:
### Description

Fatal fault when running test-k64k-pins.js
 
### Test Code

[test-k64f-pins.js](https://github.com/01org/zephyr.js/blob/master/tests/test-k64f-pins.js);
 
### Steps to Reproduction

1. `make BOARD=frdm_k64f JS=tests/test-k64f-pins.js` 
2. cp outdir/frdm_k64f/zephyr.bin /media/<USERNAME>/MBED/

### Actual Result

[image]

 
### Expected Result

No error thrown

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | bb11854 | FRDM-K64F | Mar 28, 2017 | Fail
 
### Additional Information

grgustaf, 2017-03-28:
@brianjjones hoping you could take a look at this, try to track down what changed to cause the problem. There are a number of broken things in K64F right now so I need some help.

qiaojingx, 2017-04-01:
Verified with commit f3d587d + #954. This issue is not reproducible.
Will close it after patch merged.

grgustaf, 2017-04-03:
Wow, three bugs with one patch. :)

qiaojingx, 2017-04-06:
Verified with commit ddcbbbe on K64F. This issue is not reproduced. Closed

closed 2017-04-06 by qiaojingx